jungmannlab/picasso #314: error in loading 3D calibration file v0.5.7

ShaulliX, 2023-02-08:
* Picasso version: 0.5.7
* Operating System: win10
* compiled version- end user

### Description

In the latest version of Picasso i can not upload a 3D calibration file in Localize (.nd2 or .tiff ), thus can not do a 3D calibration.

I uninstalled the latest version and went back to the one i had before (v0.4.5) and it worked without any problem. 
Attached below the error message i get.

Thanks for the help :)

[image]

rafalkowalewski1, 2023-02-10:
Hi

Thank you for raising the issue. When it comes to .nd2 files, they are not very well standardized and thus error-prone, we are working on improving this.

On the other hand, to use the .tiff file, simply manually change the extension to .ome.tif (instead of .tiff). This should work, let me know if that's not the case.

Bests,
Rafal

ShaulliX, 2023-02-10:
> 

Hi Rafal,

Thank you for the tip. It worked :)

Best,
Xhorxhina

closed 2023-02-12 by rafalkowalewski1