jungmannlab/picasso #326: Point Spread Function width value sx incoherent

IrrelevantNameTom, 2023-03-03:
* Picasso version: 0.5.6
* Python version: standalone version
* Operating System: windows 10

### Description

Hello,
I was looking at the values of the PSF fit width "sx" in the Filter package and most values are centered on 0.9 (pixel size 130 nm). I thought that this value should be around 1.90 - 2. Indeed it would mean that the PSF width (hence my lateral resolution) of my microscope is around 120 nm which is of course not possible. 

I am looking at DNA nanorulers with the Cy3B dye (from gattaQuant). I am imaging with a Nikon Ti E + TIRF on an Andor ixon ultra 888 camera with a 100X TIRF oil objective. 
Either the sx is the SD of the PSF and not the width, or my pixel size is wrong from the start... 
Anyone has an idea ? Thanks ! : )
By the way, I love your packages.

Tom

rafalkowalewski1, 2023-03-03:
Hey Tom, 

sx having value around 0.9 - 1.1 means that your setup is calibrated just right. sx is the sigma in the gaussian fit of that specific spot which does not equal resolution. The pixel size of 130 nm is also right - we have the exact same value in our lab. I hope this helps : )

Bests,
Rafal

IrrelevantNameTom, 2023-03-03:
Hey Rafal,

Glad to hear that ! Thanks for the quick answer : ) 

Best,
Tom

closed 2023-03-03 by IrrelevantNameTom