jungmannlab/picasso #327: .Tif loading error

CDGSU, 2023-03-17:
* Picasso version:
* Python version:
* Operating System:

### Description

I am trying to load .tif videos taken in an EMCCD camera with the Picasso: Localize software. But I am not able to do so and its showing me the following error.

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.
If you can, upload the dataset that caused the problem.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```
[image]

rafalkowalewski1, 2023-03-18:
Hi,

What is the extension of your file? Is it .ome.tif, .tif, or .tiff? Also, did you acquire the images using MicroManager?

Bests,
Rafal

CDGSU, 2023-03-18:
Thanks for the quick response.

The file extension is .tif, and we are using the Andor acquisition software.

Best

rafalkowalewski1, 2023-03-19:
This may explain why the loading does not work. Picasso was built to work primarily on datasets generated by MicroManager. One solution I may advise is to convert your file to the .raw extension, using Picasso Toraw. You can run this module using the instructions from here: https://github.com/jungmannlab/picasso#:~:text=Via%20PyPI,localize%2C%20etc. instead of "render" or "localize", type "toraw".