jungmannlab/picasso #328: Error when running 3d calibration routine

mb1069, 2023-03-20:
* Picasso version: v0.6.0
* Python version: 3.8.10
* Operating System: Ubuntu

### Description
Error when running 3d calibration routine
```
Traceback (most recent call last):
  File "/home/miguel/Projects/uni/phd/smlm_z/old/picasso/picasso/gui/localize.py", line 1720, in on_fit_finished
    zfit.calibrate_z(
  File "/home/miguel/Projects/uni/phd/smlm_z/old/picasso/picasso/zfit.py", line 48, in calibrate_z
    mean_sx = interpolate_nan(mean_sx)
  File "/home/miguel/Projects/uni/phd/smlm_z/old/picasso/picasso/zfit.py", line 22, in interpolate_nan
    data[nans] = _np.interp(x(nans), x(~nans), data[~nans])
  File "<__array_function__ internals>", line 180, in interp
  File "/home/miguel/Projects/uni/venv/lib/python3.8/site-packages/numpy/lib/function_base.py", line 1570, in interp
    return interp_func(x, xp, fp, left, right)
ValueError: array of sample points is empty
```

[bead_stack.ome.tif.zip](https://github.com/jungmannlab/picasso/files/11019271/bead_stack.ome.tif.zip)

### What I Did
Ran 3d calibration routine in localisation uI

rafalkowalewski1, 2023-03-25:
Hey Miguel, sorry for a late reply.

Do you use an astigmatic lens? Just asking since the spots look a bit unusual.
The reason why you get the bug is that Localize automatically filters out the fitted localizations whose PSF widths vary too much from the mean. The cutoff is taken from the variance of these PSF widths and in the case of your data the variance is zero - it appears you are using some simulation here? 

Theoretically, you could change your code (/home/miguel/Projects/uni/phd/smlm_z/old/picasso/picasso/zfit.py) and comment out line 41 (locs = locs[keep]). However, this will cause problems in the future when you try calibrating using realistic data.

I hope this helps, let me know if there is anything else I could assist you with.
Bests,
Rafal

mb1069, 2023-03-26:
Hi Rafal,
You are absolutely right, this is a simulated PSF with vertical astigmatism. Thanks for clarifying that, I'll add some noise into the dataset and try this again.

Cheers,
Miguel

closed 2023-03-26 by mb1069