jungmannlab/picasso #332: KeyError: 'cpu_utilization'

smurkel, 2023-04-24:
* Picasso version: 0.6.0
* Python version: 3.8
* Operating System: 64 bit Windows

### Description

In the Localize module, I get this error when I choose either 'Identify' or 'Localize', but not with just 'Fit':

[image]

However, when I activate 'Preview' in the Analyze -> Parameters menu, red boxes do appear around some particles, so it seems that the localization does work in that case. Fit does not give an error, but also nothing seems to happen - when I run 'Fit' and try to export the localizations to see if there are any, I get the following error:

[image]

It don't think it is related to the following but just in case: the way I loaded the .tif file is a bit hacky; it's a .tif that was created using Python module PIL and at first I couldn't open it, until I changed the extension to .ome.tif

I would appreciate some help!

rafalkowalewski1, 2023-04-25:
Hi

Thanks for raising the issue. We still cannot figure out why the problem occurs but here is how to solve it:

* Start by navigating to your home directory in the File Explorer (not Desktop but the directory above that).

* You need to enable showing hidden files, please follow these steps: https://support.microsoft.com/en-us/windows/show-hidden-files-0320fe58-0117-fd59-6851-9b7f9840fdb2

* Enter the ".picasso" folder and then open the "setting.yaml" file (use any text editor for that)

* Under "Localize" add a new line "  cpu_utilization: 0.8" (please mind the double space in the beginning. This should look like this: https://datashare.biochem.mpg.de/s/IMmtxtmEcnI8z9n

* Save the "settings.yaml" file.

Please let me know if that helped!
Bests,
Rafal

closed 2023-05-03 by rafalkowalewski1

smurkel, 2023-05-03:
Hi, sorry for the late response! In the meanwhile I had moved to a different PC where the software worked  without any problems (same OS, same pretty much everything, so no useful info I can give I'm afraid), and then I forgot about this post for a bit. 

I just tested your suggested solution on the original PC and it does indeed fix the problem!