jungmannlab/picasso #334: Pixel size as additional argument and batch export

agvesga, 2023-05-16:
* Picasso version: v0.6.0 
* Python version: 3.8
* Operating System: Linux

### Description

I need to input the pixel size as additional argument in CMD line. For instance:

```
python -m picasso localize /path/Sample/storm_1/ -a mle -g 1000 -bl 100 -s 0.458 -qe 0.82 -b 14      -pxs 65
```

Where ```pxs``` stand for pixel size

On the other hand, if this is possible to do in a python script, it would be great also, altough I did not manage to build a python script for batch localization yet... Is there any example?

Another question is, how can I batch filter + export several files? Could this be done with a python script or through CMD?

Thanks

rafalkowalewski1, 2023-05-17:
Hi,

The pixelsize argument can be input using the following argument: "pixelsize 130" to set it to 130 nm, although the only thing that it will change is the pixelsize value in the resulting .yaml file.
To batch process files, you need to provide the name of the folder where you store your files for localization, instead of the path to a single .tif file, please see here: https://picassosr.readthedocs.io/en/latest/cmd.html#batch-process-a-folder.
When it comes to filter, it is not supported in CMD. We will consider adding this functionality in the future.

Let me know if this helps.
Bests,
Rafal

agvesga, 2023-05-17:
Hi, 

thanks a lot for the answer. Nevertheless, when I run this:

```
python -m picasso localize /camp/project/proj-czi-vp/raw/lm/03052023_laser_power_titration/EM04208_11/Sample_1/Section_1/FOV1/storm_1/ -a mle -g 1000 -bl 100 -s 0.458 -qe 0.82 -b 9 pixelsize 65
```

I get the following 

```
usage: picasso [-h] {toraw,localize,filter,render,link,cluster_combine,cluster_combine_dist,clusterfilter,undrift,density,dbscan,hdbscan,smlm_cluster,dark,align,join,groupprops,pc,nneighbor,design,simulate,nanotron,average,average3,hdf2visp,csv2hdf,hdf2csv,server} ...
picasso: error: unrecognized arguments: pixelsize 65
```

Also do not work with ```-pixelsize```

Nevertheless, it would be of a lot of help for me to run all these batch analysis in a python script, is there any working example you have?

Thanks a lot,
Arturo

rafalkowalewski1, 2023-05-22:
Hey Arturo,

I'm sorry, I made a mistake, you should type "--pixelsize 65", i.e., with the double dash. When it comes to the examples we do not have any

agvesga, 2023-05-22:
Hi, 

It works, indeed. Out of curiosity, how many other arguments apart from the ones explained [here](https://picassosr.readthedocs.io/en/latest/cmd.html#adding-additional-arguments) and ```--pixelsize``` can I add in CMD? They may be useful for me in the future.

Thanks !

rafalkowalewski1, 2023-05-23:
Hi

Additionally, you can add arguments for 3D localization:
mf - magnification factor
zc - z calibration file path (the .yaml file with z calibration coefficients)

I hope this helps
Bests
Rafal

closed 2023-05-23 by rafalkowalewski1