jungmannlab/picasso #343: Error at Localize module launch after installing GPUFit on a pip install

cleterrier, 2023-07-26:
I tried to install Picasso with the pip install method in an Anaconda environment under WIndows 10. I did as per the readme file:

Via PyPI
Open the console/terminal and create a new conda environment: `conda create --name picasso python=3.8`
Activate the environment: `conda activate picasso`
Install Picasso package using: `pip install picassosr`
You can now run any Picasso function directly from the console/terminal by running: picasso render, picasso localize, etc.

Then I installed the latest GPUFit wheel (https://github.com/gpufit/Gpufit/releases/tag/v1.2.0, cublas version) using `pip install [PATH_TO_GPUFIT_WHEEL]`

Output was:
```
Processing c:\users\chris\christo\processing\gpufit_1.2.0_cublas_build202110131846\python\pygpufit-1.2.0-py2.py3-none-win_amd64.whl
Requirement already satisfied: NumPy>=1.0 in c:\users\chris\.conda\envs\picasso\lib\site-packages (from pyGpufit==1.2.0) (1.19.2)
Installing collected packages: pyGpufit
Successfully installed pyGpufit-1.2.0
```

It created the two folders `pygpufit` and `pyGpufit-1.2.0.dist-info` in the environment `C:\Users\chris\.conda\envs\picasso\Lib\site-packages`

Now when I launch the Localize module `picasso localize` I get the following error:
```
Traceback (most recent call last):
  File "C:\Users\chris\.conda\envs\picasso\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\chris\.conda\envs\picasso\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\chris\.conda\envs\picasso\Scripts\picasso.exe\__main__.py", line 7, in <module>
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\__main__.py", line 1658, in main
    from picasso.gui import localize
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\gui\localize.py", line 20, in <module>
    from .. import io, localize, gausslq, gaussmle, zfit, lib, CONFIG, avgroi
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\picasso\gausslq.py", line 21, in <module>
    from pygpufit import gpufit as gf
  File "C:\Users\chris\.conda\envs\picasso\lib\site-packages\pygpufit\gpufit.py", line 24, in <module>
    lib = cdll.LoadLibrary(lib_path)
  File "C:\Users\chris\.conda\envs\picasso\lib\ctypes\__init__.py", line 451, in LoadLibrary
    return self._dlltype(name)
  File "C:\Users\chris\.conda\envs\picasso\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'C:\Users\chris\.conda\envs\picasso\Lib\site-packages\pygpufit\Gpufit.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```
whereas the file at this path is indeed existing.

rafalkowalewski1, 2023-08-01:
Hi Christophe,

Apologies for the late reply. We will try solving this issue with the next Picasso release. Sorry for inconvenience.

Best wishes,
Rafal

rafalkowalewski1, 2023-09-13:
Hi Christophe,

I have looked into the problem and I noticed that if you choose the non-cublas version, everything works fine. Again, I would like to apologise for taking such a long time to reply.

Bests,
Rafal

closed 2023-09-13 by rafalkowalewski1