jungmannlab/picasso #344: Error while localizing spots with .ims

annafasoli, 2023-08-03:
* Picasso version: 0.5 and above
* Python version: N/A - using the installer
* Operating System: Windows 10

### Description

The .ims files are loading fine, and I can start running the localization step but then I get an error when all the locs are found, at the "Preparing Fit" stage.
Sometimes the error is an empty error window, sometimes it shows like below.

Version 0.4.11 works OK.

Thank you,
Anna

[image]
[image]

rafalkowalewski1, 2023-08-04:
Hi Anna,

Thank you for your message. Could you please share the file so that I could take a closer look?

Bests,
Rafal

rafalkowalewski1, 2023-08-04:
you can send it to this email: kowalewski@biochem.mpg.de

annafasoli, 2023-08-04:
Thanks, Rafal. I have shared the data with you

closed 2023-09-15 by rafalkowalewski1