jungmannlab/picasso #392: Issue with yaml constructor using Picasso Join

GolgiWhillikers, 2023-10-04:
* Picasso version: 0.6.4
* Python version: 3.10
* Operating System: Win 10

I am running into an issue with yaml constructs when nd2 files are localized in picasso 0.6.4 that did not occur in 0.6.0. Essentially, I localize several files using picasso localize cmd line function, then use picasso join in command line to combine them. This worked fine in 0.6.0. In 0.6.4, I get an error in picasso join in yaml constructors that is a line in the nd2 metadata section of the _locs.yaml file.

Here is a screenshot of the command and associated error
[image]

Here is the section of _locs.yaml that threw the error
[image]

This appears to specifically be an issue with picasso 0.6.4, I suspect stemming somewhere in the localize workflow:
-If I localize files in 0.6.0 and join in 0.6.0, this error does _not_ occur
-If I localize in 0.6.4 and join in 0.6.0, this error _does_ occur
-If I localize in 0.6.0 and join in 0.6.4, this error does _not_ occur
-If I localize in 0.6.4 and join in 0.6.4, this error _does_ occur

I I was able to "fix" this in 0.6.4 by editing line 201 (load_info function) of picasso's io.py to use the _yaml.UnsafeLoader instead of _yaml.FullLoader (taken from https://github.com/yaml/pyyaml/issues/482, but honestly I don't really know what I'm doing and just tried something that hit the keywords). However these functions are the same in both picasso io.py 0.6.0 and 0.6.4 so I suspect it's not actually the problem, and just a downstream solve. 

Here is the line of io.py that I changed to unsafeloader to "fix" the problem
[image]

I have noticed a warning about nd2 file garbage collection in picasso localize from cmd line in picasso 0.6.4 that I don't get in 0.6.0. It doesn't prevent localization, but does pop up, and could be unrelated. I see there is also a new version of the python nd2 package being used and an nd2reader - I don't know the difference with older verisons but I could see that being the source.

Here is the line in picasso 0.6.0 with no error
[image]

And here is the line in picasso 0.6.4 with the error
[image]

I have also noticed that the yaml files after picasso localize are not the same between the two picasso versions, which again makes me think it's something about the change in nd2 reading. See here for the loc and yaml files localized with both versions that I've been using.
https://drive.google.com/drive/folders/1LUiOWPHQUyaQIU-xASJTmVs8LoiMYfW4?usp=sharing

Thanks for the continued support and troubleshooting!

rafalkowalewski1, 2023-10-06:
Hello,

Thank you for raising the issue and pointing out so much detail! You are right, most likely the errors occur due to the updated dependencies and it's difficult to predict what could go wrong here. I will take a look at this in the coming days.

Bests,
Rafal

rafalkowalewski1, 2023-10-22:
Hi, I updated picasso to 0.6.5, using your method, everything should work fine now.

Bests,
Rafal

closed 2023-10-22 by rafalkowalewski1

GolgiWhillikers, 2023-10-24:
Confirmed for me that all the cmd functions worked appropriately in a fresh env using 0.6.5. Thanks for the quick update.