jungmannlab/picasso #405: Issue with localize function

Daan-Haas, 2023-11-01:
* Picasso version: 0.6.1
* Python version: 3.10.13
* Operating System: Win10

### Description

Calling picasso.localize.localize on the rawdata dataset from the git gave an error. Looking through the code, localize calls identify but doesn't pass an argument for 'box' 

### What I Did

```
params = io.load_info(infofile)
movie, info = io.load_raw(rawfile)

locs = localize.localize(movie, info, params)
```

and the error:
```
    locs = localize.localize(movie, info, params)
 line 428, in localize
    identifications = identify(movie, parameters)
TypeError: identify() missing 1 required positional argument: 'box'
```

rafalkowalewski1, 2023-11-01:
Hi,

Thanks for raising the issue. This is an old piece of code which we will solve with the next Picasso release. For now, I recommend to use Picasso localize directly from the command window (not within the python script) as described here: https://picassosr.readthedocs.io/en/latest/cmd.html#localize 

Bests,
Rafal

rafalkowalewski1, 2024-07-06:
Hello again,

Apologies for the late fix. In Picasso 0.7.0, the new function will be:

`def localize(movie, camera_info, parameters):
    identifications = identify(
        movie,
        parameters["Min. Net Gradient"],
        parameters["Box Size"],
    )
    return fit(movie, camera_info, identifications, parameters["Box Size"])`

Hope this helps, thank you for raising the issue.

Bests,
Rafal

closed 2024-07-06 by rafalkowalewski1