jungmannlab/picasso #435: hdf2csv throws error that it is missing pytables

lilbutsa, 2024-01-30:
* Picasso version: 0.6.8
* Python version: Windows install
* Operating System: Windows 10

### Description

Hi,
I was trying to run the hdf2csv function through command line in windows but it throws an error that it is missing the pytables dependency. What is the best way to overcome this? Should I try and add it manually to the Picasso program folder? 

Thanks for your help and for developing such great software!

James

### What I Did

>picasso hdf2csv Test.hdf5
  0%|                                                                                            | 0/1 [00:00<?, ?it/s]Converting Test.hdf5
  0%|                                                                                            | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "pandas\compat\_optional.py", line 132, in import_optional_dependency
  File "importlib\__init__.py", line 126, in import_module
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'tables'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "picasso_pyinstaller.py", line 9, in <module>
  File "picasso\__main__.py", line 1747, in main
  File "picasso\__main__.py", line 174, in _hdf2csv
  File "pandas\io\pytables.py", line 426, in read_hdf
  File "pandas\io\pytables.py", line 566, in __init__
  File "pandas\compat\_optional.py", line 135, in import_optional_dependency
ImportError: Missing optional dependency 'pytables'.  Use pip or conda to install pytables.
[29040] Failed to execute script 'picasso_pyinstaller' due to unhandled exception!

rafalkowalewski1, 2024-02-02:
Hi,

Thanks for raising the issue. I am not sure what may be causing it, maybe it'd be possible to use the PyPI release of picasso? https://pypi.org/project/picassosr/

Simply follow the instructions on our github page under that "Via PyPI" section: https://github.com/jungmannlab/picasso?tab=readme-ov-file#via-pypi

I hope this helps, please let me know if the issue persists.

Bests,
Rafal

rafalkowalewski1, 2024-06-05:
Hello again,

I looked at the problem again, the solution is to install the package called "pytables", either from pip or conda.

I changed the code such that this is no longer required and in the next version (0.6.10) it should work without problems.

Bests,
Rafal

closed 2024-06-05 by rafalkowalewski1

lilbutsa, 2024-06-17:
Hey Rafal,

Thanks for fixing that and keep up the great work!

James