jungmannlab/picasso #450: Suggestion: Implement different AIM drift correction algorithm (or DME)

ajasja, 2024-06-03:
I recently came across this paper (https://www.science.org/doi/10.1126/sciadv.adm7765), which presents a novel drift correction method called AIM.
[image].

It looks almost too good to be true. We are planning to test the Matlab version (https://github.com/YangLiuLab/AIM).
If it works well, would there be interest in including AIM in picasso? (I realize there is no Python version yet, but the algorithm seems quite easy to port over to Python). 

Any comments on AIM are also welcome.

rafalkowalewski1, 2024-06-05:
Hi, 

I agree, the algorithm looks great, we will consider your suggestion!

Bests,
Rafal

ajasja, 2024-06-06:
So it seems to work at least as good as RCC. Is there a good way to quantify resolution? I guess I could do a nearest neighbor analysis. 

RCC:
[image]

AIM:
[image]

The algorithm was super fast. About 2 s. RCC was about 5-10s.

I wrote a hacky matlab script 
[AIM-main.zip](https://github.com/user-attachments/files/15621776/AIM-main.zip)
to do the alignment:
```matlab
%% Example code
   % align the picasso images
clc
clear
close all
warning('off')
addpath(genpath('.'))
addpath(genpath('./AIM'))
addpath(genpath('./DME_RCC'))
addpath(genpath('./Data'))

%% Load HDF5 picasso
in_file = 'Data/C1-20240604-164932_WT.ome_locs.hdf5'
%inh5 = H5F.create (in_file, 'H5F_ACC_TRUNC', 'H5P_DEFAULT', 'H5P_DEFAULT');

%% load HDF5
locs = h5read(in_file, '/locs') 

%% align

Localizations(:,1) = double(locs.frame);
Localizations(:,2) = double(locs.x);
Localizations(:,3) = double(locs.y);

%% AIM drift correction
trackInterval = 50; % time interval for drift tracking, Unit: frames 
t_start = tic;
[LocAIM, AIM_Drift] = AIM(Localizations, trackInterval);
AIM_time = toc(t_start)

%% save output
olocs.frame = LocAIM(:, 1) 
olocs.x = LocAIM(:, 2) 
olocs.y = LocAIM(:, 3) 
olocs.lpx = locs.lpx
olocs.lpy = locs.lpy

out_name = [in_file(1:end-4) '_undrifted.hdf5']

% Not working for some reason in 2020b
% h5write(out_name, '/locs', locs)
save_picasso('.', out_name, olocs)
```

rafalkowalewski1, 2024-07-05:
AIM will be implemented in Picasso 0.7.0.

closed 2024-07-05 by rafalkowalewski1

ajasja, 2024-07-05:
Thanks! Looking forward to that.

ajasja, 2024-07-09:
Amazing, I just saw AIM is now in 0.7. Can't wait to try it out.

rafalkowalewski1, 2024-07-09:
The installer is in the making : )