jungmannlab/picasso #479: NSException in libc++abi When Running `picasso localize`

inbasperu, 2024-06-11:
### Issue Report: NSWindow Exception When Running `picasso localize`

**Python Version:** 3.10  
**Operating System:** macOS Sonoma 14.2.1  
**Device:** M3 Pro  
**Installation Method:** Installed from source

---

#### Description
When attempting to run the command `picasso localize`, I encountered an uncaught exception related to `NSWindow` initialization. The application terminates unexpectedly with the following error message:

```bash
QObject::setParent: Cannot set parent, new parent is in a different thread
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow should only be instantiated on the main thread!'
*** First throw call stack:
(
        0   CoreFoundation                      0x000000018261c570 __exceptionPreprocess + 176
        1   libobjc.A.dylib                     0x000000018210deb4 objc_exception_throw + 60
        2   CoreFoundation                      0x00000001826418bc _CFBundleGetValueForInfoKey + 0
        3   AppKit                              0x0000000185d93194 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 188
        4   AppKit                              0x0000000185eecb3c -[NSPanel _initContent:styleMask:backing:defer:contentView:] + 48
        5   AppKit                              0x0000000185d930cc -[NSWindow initWithContentRect:styleMask:backing:defer:] + 48
        6   AppKit                              0x0000000185eecaf0 -[NSPanel initWithContentRect:styleMask:backing:defer:] + 48
        7   AppKit                              0x0000000185feece0 -[NSWindow initWithContentRect:styleMask:backing:defer:screen:] + 24
        8   libqcocoa.dylib                     0x00000002ad3eaa70 qt_plugin_instance + 188964
        9   libqcocoa.dylib                     0x00000002ad3d687c qt_plugin_instance + 106544
        10  libqcocoa.dylib                     0x00000002ad3cffa4 qt_plugin_instance + 79704
        11  libqcocoa.dylib                     0x00000002ad3cfabc qt_plugin_instance + 78448
        12  QtGui                               0x0000000104290d04 _ZN14QWindowPrivate6createEby + 164
        13  QtWidgets                           0x0000000103ce0c44 _ZN14QWidgetPrivate6createEv + 1180
        14  QtWidgets                           0x0000000103cdf9ac _ZN7QWidget6createEybb + 320
        15  QtWidgets                           0x0000000103cf16a0 _ZN14QWidgetPrivate10setVisibleEb + 180
        16  QtWidgets                           0x0000000103e930f8 _ZN7QDialog10setVisibleEb + 272
        17  QtWidgets                           0x0000000103e928f4 _ZN7QDialog4execEv + 224
        18  QtWidgets                           0x0000000103ecee80 _ZN11QMessageBox11informationEP7QWidgetRK7QStringS4_6QFlagsINS_14StandardButtonEES6_ + 372
        19  QtWidgets.abi3.so                   0x0000000104fc9f68 _ZL25meth_QMessageBox_criticalP7_objectS0_S0_ + 216
        20  python3.10                          0x0000000100fa4158 cfunction_call + 60
        21  python3.10                          0x0000000101062fc4 _PyEval_EvalFrameDefault + 58924
        22  python3.10                          0x0000000100f47364 _PyFunction_Vectorcall + 548
        23  python3.10                          0x00000001010cc7c0 _PyObject_VectorcallTstate.5848 + 92
        24  python3.10                          0x00000001010cb4e8 _PyErr_PrintEx + 356
        25  QtCore.abi3.so                      0x0000000102c59af8 _Z15pyqt5_err_printv + 128
        26  sip.cpython-310-darwin.so           0x0000000101ab23ec sip_api_call_procedure_method + 164
        27  QtCore.abi3.so                      0x0000000102b3427c _ZN10sipQThread3runEv + 112
        28  QtCore                              0x00000001033ee2f4 _ZN7QThread11qt_metacallEN11QMetaObject4CallEiPPv + 1296
        29  libsystem_pthread.dylib             0x00000001824ca034 _pthread_start + 136
        30  libsystem_pthread.dylib             0x00000001824c4e3c thread_start + 8
)
libc++abi: terminating due to uncaught exception of type NSException
zsh: abort      picasso localize
```

Additionally, the following warning is displayed during shutdown:

```bash
/opt/homebrew/Caskroom/miniconda/base/envs/picasso/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 5 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
```

#### Steps to Reproduce
1. Install `picasso` from source on macOS Sonoma 14.2.1.
2. Run the command `picasso localize`.

Please let me know if any additional information or logs are required to diagnose this issue. 

Thank you for your time and assistance.

rafalkowalewski1, 2024-06-11:
Hello,

Thank you for reporting the problem. What Picasso version do you use? 

Bests,
Rafal

inbasperu, 2024-06-18:
> Hello,
> 
> Thank you for reporting the problem. What Picasso version do you use?
> 
> Bests, Rafal

I built Picasso from source. Here is the version I am using:
`-e git+https://github.com/jungmannlab/picasso.git@35c3d8eb164c3e50083744d23cf5f361b9108c5e#egg=picassosr`

rafalkowalewski1, 2024-06-19:
Can you please go to your picasso folder and find the file called __version__.py? The version number is reported there.

Bests,
Rafal

inbasperu, 2024-06-19:
> Can you please go to your picasso folder and find the file called **version**.py? The version number is reported there.
> 
> Bests, Rafal

Version number is "0.6.9"

rafalkowalewski1, 2024-07-05:
Hi,

Sorry for being inactive for some time. Could you please check if other Picasso modules work? For example, render or design?

Many thanks,
Rafal

closed 2025-02-04 by rafalkowalewski1