jungmannlab/picasso #480: Command line error - conflicting subparser

mb1069, 2024-06-18:
* Picasso version:
* Python version: 3.11
* Operating System: Linux (Ubuntu)

### Description

Running command
`picasso localize -h`

causes error:
```
Traceback (most recent call last):
  File "/home/miguel/anaconda3/envs/psf_analyzer/bin/picasso", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/miguel/anaconda3/envs/psf_analyzer/lib/python3.11/site-packages/picasso/__main__.py", line 1535, in main
    render_parser = subparsers.add_parser(
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miguel/anaconda3/envs/psf_analyzer/lib/python3.11/argparse.py", line 1197, in add_parser
    raise ArgumentError(self, _('conflicting subparser: %s') % name)
argparse.ArgumentError: argument command: conflicting subparser: render

Traceback (most recent call last):
  File "/home/miguel/Projects/PSF_analyzer/psf_analyzer/prepare_tif/main.py", line 695, in <module>
    test_picasso_exec()
  File "/home/miguel/Projects/PSF_analyzer/psf_analyzer/prepare_tif/main.py", line 53, in test_picasso_exec
    raise EnvironmentError('Picasso not found/working (see above)')
OSError: Picasso not found/working (see above)
```

The issue appears to be the following lines in __main__.py

```
     for command in ["toraw", "localize", "filter", "render"]:
         subparsers.add_parser(command)
```

### What I Did
Removing those lines allows the subparser to function correctly

rafalkowalewski1, 2024-06-19:
Hi Miguel,

I unfortunately cannot reproduce the error you reported. I download picasso from PyPI and run "picasso localize -h" and it functions correctly.

Bests,
Rafal

closed 2024-06-19 by rafalkowalewski1

mb1069, 2024-06-19:
Hi Rafal,
I just created a new conda env to test it, and could reproduce the error:

Using conda 24.1.2
```
conda create --name tmp -c conda-forge python=3.11;
conda activate tmp;
pip install picassosr;
pip freeze | grep picasso => picassosr==0.6.11
```

Kindest,
Miguel

rafalkowalewski1, 2024-06-19:
Hi again,

I think the difference between our approaches is the python version. Picasso is developed on python 3.10, when I use 3.11 I get the same error.

To be honest, I have no idea why this happens, I would've expected such basic functions as argument parsing to be stable across python releases.

I will look at this in the future, but I'd rather recommend using python 3.10.

Many thanks for pointing this out!

Bests,
Rafal