jungmannlab/picasso #482: localize via command-line + mle fitting: sx == sy

MariusGlg, 2024-06-27:
* Picasso version: 0.6.11
* Python version: 3.10.13
* Operating System: Windows

### Description

The Point Spread Function width in camera pixels is equal in x & y (sx == sy) when using the command line for localization + mle as the fitting option. This is not the case when using mle fitting in the picasso render module (see file attached)
[image]

### What I Did

```
picasso localize "filepath" -b 7 -a mle -g 150000 -r 165 122 267 227 -bl 200
```

rafalkowalewski1, 2024-06-28:
Hello Marius,

Thank you for pointing this out. I will fix this at the new update.

Bests,
Rafal

rafalkowalewski1, 2024-07-05:
Hello Marius,

I decided that the CMD version should output different sigmas for x and y (this can be useful for filtering spots that have too different values later). The update will be available next week.

Bests,
Rafal

closed 2024-07-05 by rafalkowalewski1