jungmannlab/picasso #492: Linking of localizations does not work in v0.7.0

CKSpahn, 2024-08-07:
* Picasso version: v0.7.0 (executable installation and python distribution)
* Python version: 3.10
* Operating System: Windows 11 Enterprise

### Description

Dear Picasso Team,

first of all thanks for the great software and the constant updates. 
When using the most recent version (v0.7.0), I experienced that linking of localization is not working properly. During linking (both executable installation and python distribution), I get the following error message:

[image]

Best wishes,
Chris

closed 2024-08-19 by rafalkowalewski1

alicia-cuber, 2024-10-08:
Hi, 
I am having the same problem (also in v0.7.0), I noticed that there has a been a new release and I was wondering if this has been resolved in the new version. If so, I will switch over. 
Best, 
Alicia

rafalkowalewski1, 2024-10-09:
Hi, the issue was resolved in 0.7.3.

Bests,
Rafal