jungmannlab/picasso #530: n2d files don't open with latest Picasso version 0.8.1

agvesga, 2025-07-21:
Hi, 

just letting you know that I get the following error while opening a nd2 file with the latest version: "AttributeError: 'ND2Reader' object has no attribute 'info'". And it can't open the image. It works with tif files.

Downgraded to vertion 0.7.5. and I can open the same nd2 file without error.

Best,

mmarinriera, 2025-07-30:
Hi,

We are also running into the same error with 0.8.1
`AttributeError: 'ND2Reader' object has no attribute 'info'`

Rolling back to 0.8.0 solves the issue.
Cheers,

rafalkowalewski1, 2025-07-31:
Hello both,

I am very sorry, it was my mistake, I will fix the bug in the next release (probably end of this week/next week)

Best,
Rafal

closed 2025-07-31 by rafalkowalewski1